Comet-Robotics/chessbots-embedded #4: Determine robot runtime on battery

jasonappah, 2025-02-17:
#### Task Description: 

We need to test how long a ChessBot would last on battery before dying. We need to know this as we plan to order more batteries soon and need to figure out what needs to be ordered for each bot.

#### Implementation Steps: 

- Determine a reasonable method to test the battery life in a setup that closely matches realistic real-world use of a ChessBot (ex: motors, IR blaster, wi-fi and messaging are all in use).
- Discuss the testing plan with the PM to ensure this is an agreeable method
- Execute said test, ensuring batteries are fully charged beforehand
- Document results in SharePoint

#### Additional Notes and Resources: 

#### Expected Timeline: 

We should have data by the end of the meeting on 3/3.

---

If you have any questions while working on this task, don't hesitate to do any one of the following:

-   Ask a lead at a meeting
-   Post in Discord
-   Reply to this issue

ymmot239, 2025-03-03:
Active runtime on the old batteries with motors and ir blasters: 1:29:25

ymmot239, 2025-03-24:
Active runtime on the new batteries with motors and ir blasters: 1:45:00 roughly

closed 2025-03-25 by xXDMOGXx