Comet-Robotics/chessbots-embedded #9: Test the encoders on all the bots

xXDMOGXx, 2025-02-24:
#### Task Description: 
We have already found out that some of our encoders are broken, and we've only tested 4 bots so far. We need to test the encoders on all of the bots so we know how many we need to order.

#### Implementation Steps: 

- Upload the main code with encoder testing enabled to an ESP
- Connect the ESP to a working PCB, and use that PCB to test each bot
- You can do this by plugging in the encoders to the encoder spots of the pcb
- You then turn the pcb on, and spin the motors by hand. If the encoder values in the log look increment and decrement normally, the encoder works
- Mark down the bot number and the encoder side for any encoders that don't work

#### Additional Notes and Resources: 

#### Expected Timeline: 
This should hopefully not take too long once you get the hang of it. It can be completed in 1-2 meetings

---

If you have any questions while working on this task, don't hesitate to do any one of the following:

-   Ask a lead at a meeting
-   Post in Discord
-   Reply to this issue

PamPamPanda22, 2025-02-25:
# Testing all encoders issue

Left Encoder of #3  and #15 do not work

#17 - WORK

#9 - WORK

#4 - WORK

#5 - WORK

#14 - WORK

#30 - WORK

#1 - WORK

#3 - LEFT NO WORK

#27 - WORK

#21 - WORK

#31 - WORK

#16 - WORK

#28 - WORK

#29 - WORK

#6 - WORK

#12 - WORK

#25 - WORK

#26 - WORK

#11 - WORK

#18 - WORK

#7 - WORK

#23 - WORK

#8 - WORK

#24 - WORK

#19 - WORK

#30 - WORK

#15 - LEFT NO WORK

xXDMOGXx, 2025-02-26:
I sorted the comment, which revealed some missing and a repeated bots. Could y'all look back over bots 2, 10, 13, 20, and 22? I'm gonna assume bot 30 works, just edit your comment and remove the repeat of it if you know for sure

PamPamPanda22, 2025-03-25:
I have checked #20 and both encoders work.

xXDMOGXx, 2025-03-25:
Thanks! We can close this out now. If you're a member, you should be able to access the list of working parts here https://cometmail.sharepoint.com/:x:/r/sites/UTDCometRobotics/_layouts/15/Doc.aspx

(Please let me know if this link doesn't work)

closed 2025-03-25 by xXDMOGXx