Comet-Robotics/chessbots-embedded #10: Test how straight the average bot drives

xXDMOGXx, 2025-02-25:
#### Task Description: 
Before we work on PID, it's a good idea to get a baseline of the average drift of a bot. We can do this by starting it at a specified point, running it for a set distance, then measuring the distance/angle from where it should be.

#### Implementation Steps: 

- Find a place to run the tests, and determine a good distance for the bots to run for the test
- Take 3 bots and upload the testing code to them (Just make both motors run the same speed for a set amount of time)
- Run them one by one. Start each one at the same starting point, and run them the same distance. Do this test 3-5 times for each bot.
- Note down the measurements for each run
- After all the runs are done, post them as a comment here, and also post the average of all of them

#### Additional Notes and Resources: 

#### Expected Timeline: 
This should hopefully be able to be done in a singe meeting

---

If you have any questions while working on this task, don't hesitate to do any one of the following:

-   Ask a lead at a meeting
-   Post in Discord
-   Reply to this issue

karmasoh000, 2025-03-09:
Bot 15
Trial 1 :     
Distance from starting point to ending point: 43 inches
Drift distance : 5 inches
Drift Angle: 6.63 degrees

Trial 2: 
Distance from starting point to ending point: 45 inches
Drift distance : 2 inches
Drift Angle: 2.54 degrees

Trial 3:
Distance from starting point to ending point: 43 inches
Drift distance : 0.5 inches
Drift Angle: 0.66 degrees

Trial 4:
Distance from starting point to ending point: 42 inches
Drift distance : 5 inches
Drift Angle: 6.789 degrees

Trial 5:
Distance from starting point to ending point: 42 inches
Drift distance : 3 inches
Drift Angle: 4.086 degrees

Average Angular Drift: 4.141 degrees

xXDMOGXx, 2025-03-09:
For now, there aren't enough working bots to test more, so this will do for now

closed 2025-03-09 by xXDMOGXx